rerun-io/rerun #5906: Remove our custom CI docker (`ci_docker`)

emilk, 2024-04-11:
We use it in fewer and fewer places, as it takes over a minute to initialize on the CI.

Having it makes it a lot more difficult to bump rust MSRV, because you need to generate a new Docker file, which you need Linux to do.

abey79, 2024-04-11:
For this, we must figure out an alternative way to build artefacts with a decent min glibc version. I've looked into that when adding support for linux aarch64, and couldn't find an alternative (which is why Jeremy jumped in and made that image arm-compatible).

emilk, 2024-04-11:
Instead of removing `ci_docker` completely, maybe we can simplify it down to its base bones (remove everything Rust from it, for instance) and do more "live" on the CI (hopefully with good caching)

jleibs, 2024-04-11:
> For this, we must figure out an alternative way to build artefacts with a decent min glibc version.

The conda build environment manages to do this (actually with a more favorable glibc -- 2.29 instead of 2.31), though it may still be dependent upon a stripped down docker container.

In an ideal world we would just use an off-the-shelf docker container, such as one of the ones provided by the manylinux project.

closed 2024-04-12 by emilk