rerun-io/rerun #5933: Pytorch lightning logger

genemerewether, 2024-04-11:
Pytorch lightning has a Logger abstraction, through which they interface with Wandb, CometML, etc.
https://lightning.ai/docs/pytorch/stable/extensions/logging.html

Any plans for something similar wrapping the Rerun Python SDK?

jleibs, 2024-04-11:
Sounds like a great idea!

alexanderswerdlow, 2024-05-28:
Closely related, it would be great it there was seamless support for embedding HTML with providers such as Wandb. Most of the value in rerun seems difficult to express directly through a lightning logger and this would also support more users (coming from a former lightning user).

From what I can tell, this is already possible with a bit of work but could be made more seamless—and certainly more efficient.

alexanderswerdlow, 2024-06-03:
@nikolausWest I didn't see it clearly in the docs, but after taking a quick look at the notebook export code, I found `rr.notebook.as_html()`! It'd be great if there was a full-screen option [instead of having to specify a fixed H/W] but it works great other than that.

It's really useful to be able to share visualizations, so I think calling this out in the docs would be helpful. On a similar note, as a new user, the whole connect/serve/save is not very intuitive, like how a viewer runs its own server by default, that two ports are needed for rr.serve(), etc, so maybe a single docs page with all the connection details would be clearer. Overall though, love what you guys have built, it's a real breath of fresh air! 

For completeness:
```
html = rr.notebook.as_html()
with open("rerun.html", "w") as f:
    f.write(html)
```

nikolausWest, 2024-06-04:
@alexanderswerdlow I hear you on the docs thing for sure.

Btw, just to make sure I understand what you're doing here, you're logging the html file as an artifact to W&B?

alexanderswerdlow, 2024-06-04:
@nikolausWest Not sure if it's technically an artifact (maybe it creates one under the hood?) but either way, a simple `wand.log({'data': wandb.Html(html)})` works!, although in the snippet above I just save to file to share separately [not through wandb].