rerun-io/rerun #5967: The 1M index entries problem

teh-cmc, 2024-04-12:
## Context

Every `log` call in Rerun results in a `DataRow`: Rerun's atomic unit of change.

`DataRow`s are indexed by the datastore, and that index ultimately powers all the row/time-based features in Rerun, such as latest-at & (visible-time-)range queries, change detection, the time panel and time cursor, garbage collection, etc.

Each entry in this index (i.e. each row) carries of a bunch of metadata:
- A `RowId`, that uniquely identifies the associated row and defines its global order (!).
- A `TimePoint`: a collection of timeline names and timestamps (generally around 3 of each in the common case).
- A bunch of `DataCell`s, each of which carry some metadata of their own:
  - A schema (as part of the underlying arrow `Array`), so that we can know whether the cell is:
    - the expected datatype (we're done)
    - a datatype that needs to be converted to the expected datatype (soft promise)
    - a promise payload that will have to be resolved (hard promise)
  - Offsets into the shared arrow buffers.
- Other stuff that I'm omitting because it's going away as part of #5303.

We also maintain a smaller, but still significant, reverse index, for when we need to fetch information about a specific `RowId`.
Finally, in addition to these primary indices maintained by the datastore itself, many pieces of the app also maintain their own secondary indices via our change detection system (e.g. all the time panel stuff).

All of this easily adds up to a few hundred bytes of overhead per index entry.
Whether that becomes a problem or not from a memory standpoint depends on the amount of data that the user actually stores in each of these index entries.

For something like e.g. a point cloud or an image, the indexing overhead is dwarfed by the actual data, and everything is fine (for the most part).
For something like a scalar, the indexing overhead is orders or magnitude larger than the data itself, which leads to situations like this one (#5904), where 20MiB of scalar data in a parquet file somehow ends up with 20GiB of memory footprint once logged.

Unfortunately the overhead issues don't stop at the memory footprint.

All row/time-based subsystems in the app must spend compute time to deal with these indices, which slows down all aspects of the viewer (including ingestion!) e.g.:
- The time panel has to spend CPU time every frame to render all these points on the timeline.
- The datastore and other secondary indices have to update their indices that much more frequently.
- The GC has to go through that many more rows.
- Etc.

And, of course, the infamous worst of all: range queries that span a lot of indices will have to unpack as many rows, which quickly becomes prohibitively expensive and therefore requires a lot of complex caching machinery to pull off (including the dreaded deserialization cache!).
Things once again get even worse from an overhead perspective if it turns out that the contents of the row was actually one single float value all along...

Having too many index entries / rows wrecks everything: from logging, to ingestion, to visualization and all the way to the memory footprint.
Given this, here's the 1M$ question: **how can Rerun help users index less rows while still somehow meeting all their needs without any (of very little) added complexity?**

Whichever solution we come up with, it should work both with static known-ahead-of-time datasets (e.g. logging a scalar plot from parquet file) and real-time datasets (e.g. an high-frequency IMU).

## Things that either wouldn't help, or not enough

### Temporal batches

- https://github.com/rerun-io/rerun/issues/4184

Temporal batches provide a huge DX and performance gain _on the logging side_, but once the data gets exploded back into rows and indexed into the store, all the problems stated above still apply.
In fact, temporal batches are just way to make it easier to create _more_ index entries!

Temporal batches also require the user to know their dataset ahead of time, while we want a solution that works both for AOT and RT use cases.

We do want temporal batches for other reasons, but they won't help with this issue.

### Optimizing the memory footprint of index entries

Some possible avenues on that front:
- Get rid of the `log_tick` timeline.
- Try and optimize the way schema information is stored and shared across rows (custom implementation of `arrow::Array`?).
- Try and optimize the memory footprint of `RowId` (varints?!).
- Others?

While those would definitely be welcomed improvements, I don't see that kind of optimization getting us anywhere near a viable memory footprint for scalar use cases.
This also only helps with the memory footprint, compute is still as much as an issue as before.

### Others?

???

---

Related:
- #5904
- #4184
- #5622
- #5303

jleibs, 2024-04-12:
> Temporal batches provide a huge DX and performance gain on the logging side, but once the data gets exploded back into rows and indexed into the store, all the problems stated above still apply.

This is only true so long a we continue to explode things back out into separate rows on ingest. It's not articulated as such in the temporal-batch issue, but if the store itself had end-to-end support for temporal batches (admittedly a lot of work) I believe this would solve many of the issues for real-world happy-path use-cases.

By happy-path, I mean if the temporal batch is pre-sorted on the same timeline as your range query and potentially even non-overlapping with other temporal batches. In this case a store that worked natively with temporal batches should only need to carry overhead of O(batches) rather than O(scalars) and be able to just return direct scalar slices from the original arrow array without ever having needed to explode these batches into separate rows in the first place.  I believe this constraint would be acceptable to most users that are really pushing the envelope of large scalar streams.

We don't need to make every possible edge-case fast. We need to provide some means for users to follow a happy-path to make a specific constrained edge-cases fast.

teh-cmc, 2024-04-16:
> Having too many index entries / rows wrecks everything: from logging, to ingestion, to visualization and all the way to the memory footprint.
> Given this, here's the 1M$ question: **how can Rerun [...]

I also forgot to mention that the solution should fit nicely with promises too.
If one has a million promises and blindly creates 1 million index entries to store them, one is doomed before even resolving anything.

emilk, 2024-04-19:
Looking a bit closer on the storage overhead:

In the `DataTable`, for each timeline we bucket the data of a component path into a `DataCellColumn`.

Logically this is a `Vec<Option<Vec<T>>>`: for each row, you optionally have a vector of values (one for each instance).
For scalar plotting this would effectively be something like `[[0.1], [0.2], [0.3], …]`.

The actual implementation of this is

```rs
pub struct DataCellColumn(pub DataCellOptVec);
pub type DataCellOptVec = VecDeque<Option<DataCell>>;

// 8 bytes
pub struct DataCell {
    // An Arc so we can share the same `DataCellInner` over different timelines,
    // e.g. the same set of point cloud colors.
    pub inner: Arc<DataCellInner>,
}

// 48 bytes
pub struct DataCellInner {
    name: ComponentName,

    size_bytes: u64,

    // Internally, this is most likely a slice of another, larger array (one microbatch per LogMsg),
    // so there is another Arc, offset, and indirection. Oh, and a DataType.
    values: Box<dyn arrow2::array::Array>,
}
```

---

Obviously this is not the best way to store small things like scalars.
We should be able to represent the entire `DataCellColumn` with one `DataType` and two arrow buffers (the flattened data plus the offsets into it).

emilk, 2024-04-22:
After a long meeting, we came up with and decided on Proposal B from above.

The main benefit of it is to drastically reduce the cost of ingestion, as well as the RAM overhead, but at the cost of slower queries (which will be amortized by a query cache).

The proposal is to arrange the data into _dense chunks_ (similar to e.g. Parquet).
Each chunk contains N rows of data for a single Entity.

* Each [temporal batch](https://github.com/rerun-io/rerun/issues/4184) becomes a single chunk
* The SDK-side batcher outputs chunks (e.g. every 8ms)
* The `DataStore` contains an index of chunks

During ingestion, each chunk is just inserted straight into the `DataStore`, which keeps a per-timeline index of the chunks.
The chunks can have overlapping time ranges. Each chunk can also be unsorted. This is what makes queries slower.

Because chunks can overlap in time, we need to use something like a [interval tree](https://en.wikipedia.org/wiki/Interval_tree) to index them. We should still have `O(log N)` insertions and queries into this tree.

In the future we can add compaction and chunk splitting to the data store in order to optimize storage and/or lookups, but we don't need it for the first iteration.

The SDK-side batcher appends data to a chunk until one of the conditions is met:

* The chunk is old enough (e.g. 8ms, our current default)
* Its get too large
* different number of timelines
* change of datatype for any of the components
* if it is sorted and larger than 1M rows
* if it is unsorted and larger than 1k rows

```rs
struct DataStore {
    entity_streams: Map<EntityPath, EntityStream>
}

struct EntityStream {
    /// Each chunk is found at most once on each timeline.
    timelines: Map<TimeLine, IntervalTree<TimeInt, ChunkId>>,
    
    chunks: Map<ChunkId, Arc<Chunk>>,
}

/// Dense storage of N rows of data for a specific EntityPath
struct Chunk {
    // Always sorted.
    row_ids: Vec<RowId>,

    // The time columns.
    timelines: Map<TimeLine, BucketTimeline>,

    /// Each `Box<dyn Array>` is effectively a `Vec<Option<Vec<T>>>` backed as raw Arrow data.
    /// The outer option here is so that we can log only `Position` one time, but not a `Color`.
    components: BTreeMap<ComponentName, Box<dyn Array>>,
}

struct BucketTimeline {
    times: Vec<TimeInt>,
    
    /// Is [`Self::times`] sorted?
    sorted: bool,
    
    /// (min, max) of [`Self::times`]
    range: TImeRange,
    
    // Future improvement: keep a shuffle-index if unsorted
}
```

abey79, 2024-07-09:
Internal slack thread containing real-world, third-party data and ingestion script that will be made _much_ greater when this issue is resolved:

https://rerunio.slack.com/archives/C040X6J573Q/p1720458010590199

emilk, 2024-08-05:
This is likely done, but needs some testing to verify.

emilk, 2024-08-08:
This is now all implemented, and will be available in 0.18.

Logging many scalars is now orders of magnitudes faster and less memory intensive

closed 2024-08-08 by emilk