rerun-io/rerun #5968: Should fall back to gl on native when vulkan fails

jleibs, 2024-04-12:
Linux environments, in particular inside of Docker frequently have issues with vulkan, but the gl backend works much more frequently.

Currently we get a cryptic crash:
[image]

Would be nice to instead fall back to gl.

Wumpf, 2024-04-13:
Somewhat familiar with this crash. I fixed it for the upcoming wgpu version (can't be backported since there's an api breakage). That fix will allow us to react to this situation which boils down to "for unclear reasons any kind of surface creation fails". Would be nice to sit together and figure out where this comes from so we can fix it in wgpu - lots of people are hitting similar stuff.

Wumpf, 2024-04-30:
I got a fix for this into wgpu by now  and it landed with 0.20. We have to upgrade our stack now and then retry

Wumpf, 2024-05-30:
oh, it looks like we're only allowing GL when it's picked explicitly, I believed this was done automatically:
https://github.com/rerun-io/rerun/blob/main/crates/re_renderer/src/config.rs#L235

That would explain the previously described behavior perfectly well. We still should be able to handle the no-adapter-at-all case better though!

closed 2024-06-17 by Wumpf