rerun-io/rerun #5984: Reset Blueprint buttons in tooltip

karimo87, 2024-04-16:
**Describe the bug**

When hovering Application ID in Blueprint panel I get two buttons in the tooltip.

**To Reproduce**
Steps to reproduce the behavior:
See video

**Expected behavior**

No buttons, or a drop down that lets me click the buttons.

**Screenshots**

https://github.com/rerun-io/rerun/assets/6584154/6c664785-0948-4f06-8025-e83bbe300b88

**Backtrace**

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.6 

**Rerun version**

15.1

emilk, 2024-04-16:
One way to solve this is to remove all buttons from tooltips.

Another way is to actually add support for clicking buttons in tooltips, by keeping tooltips around when you hover them.

emilk, 2024-07-05:
These buttons are gone.
Also, since egui 0.28 we now support actually clicking buttons in tooltips.

closed 2024-07-05 by emilk