rerun-io/rerun #6010: Configurable order of datastreams in Visualizer

rgolovanov, 2024-04-17:
Currently, visualized data has fixed order in the Visualizer. For example, when draw multiple curves on same 2d plot it is impossible to change the order - this makes the analysis of data difficult and requires toggle visibility of overlapped curves.

See https://www.rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.15.1%2Fexamples%2Fplots.rrd
It is impossible to change order of `sin` and `cos`:
[image]

Wumpf, 2024-04-17:
This bottoms out to supporting the `DrawOrder` component for time series elements 👍

rgolovanov, 2024-04-17:
@Wumpf it is more about dynamic draw order control via UI as we may not know which order is better at the code level.
I think same proposal shall be considered for 2D space view:
https://www.rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.15.1%2Fexamples%2Fhuman_pose_tracking.rrd
[image]

Wumpf, 2024-04-17:
Makes sense. I also think that draw order component (like any other component!) should be a thing one can change from the ui! But good point that this is the main important thing here and likely needs special consideration on how that's shown 👍