rerun-io/rerun #6011: Unpack rrd file into folders/images/json

rgolovanov, 2024-04-17:
Currently the only one way to see data stored in rrd file is to open it in Visualizer. 
It might be useful to be able to extract content into raw individual files.

nbkn865, 2024-09-27:
Hello, I also am trying to extract data from an .rrd file, specifically RGB and other images.  Is it true that the SDK API doesn't have this functionality yet?

Wumpf, 2024-09-28:
@nbkn865  There's ongoing work to expose APIs to access the datastore/loaded-rrd files which will land in the next release!

Specifically for anything logged as images or a blob the viewer already has an option to save those when selected (look for "save blob").

Wumpf, 2024-10-22:
0.19 supports now reading out the data store which allows to extract the data in code and store it into different files as needed.
What's now missing is to offer this out of the box
* without code
* with more utilities on the dataframe, e.g. something that extracts image files from an rrd for you

that said, it's a bit open ended at this point since there's ofc no clear definition of "generate files from this rrd".
@rgolovanov can you describe concrete usecases you had in mind?

Wumpf, 2024-11-06:
@nbkn865 since those are [`EncodedImage`](https://rerun.io/docs/reference/types/archetypes/encoded_image), i.e. jpeg blobs, size information is part of the jpeg definition
It's a different story with [`Image`](https://rerun.io/docs/reference/types/archetypes/image) which is actual "raw" pixel data which always has to be accompanied  by an `ImageFormat` which contains size among other things.

nbkn865, 2024-11-19:
@Wumpf, can you tell me how to extract the image from an EncodedImage object?  I looked at the repo and looked at the contents of the class but can't seem to find a method for extracting the image from EncodedImage.

Wumpf, 2024-11-19:
As I was trying to point out earlier, you're already looking at the image data - `Blob` is the raw jpeg data that was fed in, so you should save that out as a jpeg file or pass it to a jpeg decoding library if you want to convert it to rgb bytes

nbkn865, 2024-11-19:
Ah, I see.  I can try out that path in the future.

In the meantime, my colleague worked out how to do it, and we have 2 ways of doing so, 1 using pandas (much faster), the other using Python objects and numpy (much slower way).

I'll put the snippets here in case others have similar questions.

Pandas way:
```
import numpy as np
import io

from PIL import Image
import rerun as rr

# load the recording
recording = rr.dataframe.load_recording("my_data.rrd")

# Get camera contents
image_view = recording.view(index="capture_time", contents="/my_rgb_camera")
table = image_view.select().read_all()

# Load to pandas
df = table.to_pandas()

# Convert Blob to bytes
img_blob = df["/my_rgb_camera:Blob"].explode()
img_bytes = img_blob.apply(np.ndarray.tobytes)`

# Convert bytes to images
def bytes_to_image(bytes):
    return Image.open(io.BytesIO(bytes))

# List of images
img_series = img_bytes.apply(bytes_to_image)
```

Using Python objects and numpy:
```
# Instead of load to pandas, do these steps

# Get image Blobs
images = table.select(["/my_rgb_camera:Blob"])

# Extract Blobs, note that Blob is a list of nested lists
image_blob = images.to_pydict().get("/my_rgb_camera:Blob")

# Convert image blob list to bytes to image
image_list = []

for i in range(len(image_blob)):
    image_blob_list = image_blob[i][0]  # Un-nest the list

    image_bytes = np.array(
        image_blob_list, dtype=np.uint8
    ).tobytes()  # Must use dtype np.uint8 for tobytes() to work correctly

    image = Image.open(io.BytesIO(image_bytes))

    image_list.append(image)
```