rerun-io/rerun #6031: Function for adding multiple points to time series at once

Famok, 2024-04-18:
Hi there,

I have timeseries recordings with >1M samples (recorded with >100 kHz) and would like to display them along my videos for the same time interval. Rerun seems to be a great choice for the general idea.

Right now I have to add each point by its own using something like
```
for i in range(10):
    rr.set_time_sequence("frame_idx", i)
    rr.set_time_seconds("sensor_time", i)
    rr.log("stuff/points", rr.Scalar(i))
```
which becomes very slow above a few thousand values.

I've tried using tensors or line segments, but it doesn't seem to be possible to display them as a timeseries.

It would be great, if one could log a batch of times and values at once or use linesegments or tensors for that.

Btw.: I'm using python...

teh-cmc, 2024-04-18:
Hey, I'll refer you to my answer in a similar thread: https://github.com/rerun-io/rerun/issues/5904#issuecomment-2049057005
> `rr.Scalar` isn't really designed for large scalar series, rather it is designed for real-time use cases that are mostly concerned with the last few minutes of data (using the "Visible time range" feature). The reason for this is that `rr.Scalar` has access to all the fancy time-related features in Rerun, and those features require a lot of extra memory and CPU overhead to be made possible.
> 
> We're always working on improving the performance of `rr.Scalar`, but it fundamentally isn't compatible with large series with millions of points.
> 
> We do have an open proposal for a `ScalarChart` though, which would be a more feature-limited version of `Scalar`, with basically no memory nor CPU overhead (please add a 👍 if you're interested in something like this, it helps us prioritize!):
> 
> * [Introducing `ScalarChart` in place of `BarChart`? #5622](https://github.com/rerun-io/rerun/issues/5622)

So, unfortunately no, we don't quite support large scalar series like this yet. We are actively looking into different ways of solving that problem in a sustainable way though. I expect we will have some updates soon :tm:.

Closing as duplicate of #5904!

closed 2024-04-18 by teh-cmc

teh-cmc, 2024-04-18:
Some other pointers that could interest you:
- https://github.com/rerun-io/rerun/issues/5967
- https://github.com/rerun-io/rerun/issues/4184