rerun-io/rerun #6034: Split window across multiple monitors

genemerewether, 2024-04-18:
Not sure if Rerun's windowing system supports this, but could we split the app into multiple windows (for use with multi-monitor / multi-display)?

emilk, 2024-04-18:
`eframe` supports it yes, but we haven't implemented it for Rerun yet. It needs some design to figure out what it would mean (e.g. would be have multiple selection panels? Or have one main Rerun window, and then just pull out separate space views to their own windows?)

alik-git, 2024-12-23:
what if I just want to have two different rerun viewers open? right now if a rerun viewer is already open, and I type "rerun" in my terminal, it connects to the already open viewer. but what if I just want another completely separate one running? is there a command for that?

Wumpf, 2024-12-28:
> is there a command for that?

yes, you have to choose a different port than the default port to listen to. `rerun`  listens to `9876` by default and if it sees a rerun process already listening to this port it can't open a new viewer. You can explicitly set the port via `rerun --port 1234`