rerun-io/rerun #6052: Add script to upload .rrd for easy sharing

emilk, 2024-04-20:
Something like:

```
> pixi run upload-rrd my_rrd.rrd
Uploaded to https://some-gcs-bucket/my_rrd.rrd
View it with https://app.rerun.io/version/0.15.1?url=https://some-gcs-bucket/my_rrd.rrd
```

This will be very useful when sharing examples.

teh-cmc, 2024-04-23:
Actually I just checked and `rerun print recording.rrd` prints everything we know about the recording... except the Rerun version 🤦

emilk, 2024-04-29:
There is a draft in https://github.com/rerun-io/rerun/pull/6111 but needs some GCS magic in order to set up a HTTP endpoint for the new GCS bucket

emilk, 2024-04-30:
Next problem is CORS:

> Access to fetch at 'https://static.rerun.io/rrd/0.15.1/rgbd_3f2ebd882270ce75cec25022152f468194c8c4ef.rrd' from origin 'https://www.rerun.io' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.

commit 9c42ba7f (2024-04-30): Add script to upload an .rrd file (#6111)

### What
* Closes #6052

Example:

```
pixi run upload-rrd ../rgbd.rrd --version 0.15.1
```

Result:
https://www.rerun.io/viewer/version/0.15.1?url=https%3A%2F%2Fstatic.rerun.io%2Frrd%2F0.15.1%2Frgbd_3f2ebd882270ce75cec25022152f468194c8c4ef.rrd

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if ap [... cut]
closed 2024-04-30 by emilk