rerun-io/rerun #6056: Add numeric labels into boxes2D/Boxes3D

haixuanTao, 2024-04-21:
Currently https://docs.rs/rerun/latest/rerun/struct.Boxes3D.html definition only allows for textual labels.

Could be good to have numerical labels as in the case of Deep Learning output, additional outputs are most of the time numeric in the case of:
- Confidence,
- tracking id 
- ...

Could be fun to be able to add those labels into rerun which would also enables in the future conditional coloring and filtering.

Not sure if this is planned and if there is a tracking issue for those items but couldn't find any in the issue page.

Wumpf, 2024-04-23:
@haixuanTao sounds like you're looking for `AnnotationContext`? You can already define class ids and then specify colors and labels there https://www.rerun.io/docs/concepts/annotation-context

Anything you're missing from this?

haixuanTao, 2024-04-23:
But, I feel like it would be on the class Id level but, in the case of a confidence level of a prediction at an instant t, it would have to be on the Boxes level IMO.

Or maybe I get the AnnotationContext wrong

Wumpf, 2024-04-23:
Not sure what you mean with "be on the class id level".

How annotation context works it that you log class ids on your boxes (instead of labels and colors) and at any point further up in the hierarchy you can define what these ids mean in terms of labels & colors. Each of these you can change freely over time.

Still not entirely following what you're trying to achieve. Is it that you'd like to have other properties (like color) derived from [custom components](https://www.rerun.io/docs/howto/extend/custom-data) using some predefined rules?

haixuanTao, 2024-04-24:
Ah ok, that makes sense.

So basically, class id should be used as unique identifiers of each boxes 2D if I want to annotate one bounding box with a confidence for exemple.

I thought class id was something shared usually shared among many boxes 2D.

Thanks for taking the time.

I can close this issue

closed 2024-04-24 by haixuanTao

jleibs, 2024-04-24:
Just to clarify / summarize a bit @haixuanTao I think your understanding is correct.

Class_ids w/ annotation context let you apply discrete numeric labels to entities _and_ let you apply derived styling (colors + text labels) based on those ids. This primarily matches the standard classificatoin/segmentation use-cases (without confidence). The main value is when you will be re-using the class, since it gives you a way of keeping the styling of that class synchronized across your entities.

Class-ids don't really let you do something useful with confidence. You could always kind of "abuse" the system with discrete classes for high / mid / low confidence, but I don't think it's well aligned with convention.

As @Wumpf noted, the best way of handling additional data that's not part of the entity model, such as confidence is to just add an additional user-defined component for it, exactly as is done in the custom-data example: https://www.rerun.io/docs/howto/extend/custom-data

If you do that you will be able to see the confidence information when you select or hover the entity in the UI. We don't yet support dynamic styling based on custom user-data, but it's something that is definitely planned as part of our longer-term roadmap. 

I hope that helps.