rerun-io/rerun #6072: Renderer randomly fails to handle a `Mesh3D` changing its size

kpreid, 2024-04-23:
**Describe the bug**
If a `Mesh3D` entity, which has indices, is repeatedly logged with a growing size, then the viewer will occasionally, for certain frames (random per run but reproducible within one), not render the mesh, and produce a log message such as:

`re_space_view_spatial::mesh_cache] Failed to load mesh "/mesh": Invalid mesh given as input: Index 81 was out of bounds for 81 vertex positions -> Index 81 was out of bounds for 81 vertex positions`

**To Reproduce**
Steps to reproduce the behavior:
1.  Run this program to generate data:

    ```rust
    use rerun::external::glam::UVec3;

    fn main() -> Result<(), Box<dyn std::error::Error>> {
        let rec = rerun::RecordingStreamBuilder::new("mesh_flicker_bug").spawn()?;

        let frame_count = 360;
        for ft in 0..frame_count {
            let triangle_count = ft;

            let positions = (0..triangle_count + 2).map(|i| {
                let r = if i % 2 == 0 { 2. } else { 3. };
                let theta = ((i * 4) as f32).to_radians();
                rerun::Position3D(rerun::Vec3D([
                    r * theta.sin(),
                    r * theta.cos(),
                    (i as f32) * 0.05,
                ]))
            });

            let index_tris = (0..triangle_count).map(|t| UVec3::new(t, t + 1, t + 2));

            rec.log(
                "mesh",
                &rerun::Mesh3D::new(positions)
                    .with_mesh_properties(rerun::MeshProperties::from_triangle_indices(index_tris)),
            )?;
        }

        Ok(())
    }
    ```
2.  Drag the cursor through the timeline.
3.  Observe flickering.

**Expected behavior**
The mesh should not randomly disappear.

**Screenshots**

https://github.com/rerun-io/rerun/assets/779501/3313083e-b16b-41bc-b43b-739d5ff6487c

Note: without screen recording active, just a few frames flickered and errored. The recording makes it look deterministic, but it is not.

**Desktop (please complete the following information):**
 - OS: macOS 14.4.1

**Rerun version**
`rerun-cli 0.16.0-alpha.1+dev [rustc 1.76.0 (07dca489a 2024-02-04), LLVM 17.0.6] x86_64-apple-darwin (debug)`
`rerun-cli 0.14.1 [rustc 1.77.1 (7cf61ebde 2024-03-27), LLVM 17.0.6] x86_64-apple-darwin`

teh-cmc, 2024-04-23:
I'm curious to see if that still happens on the new data APIs. I'll have a look later when I'm back on these branches.

teh-cmc, 2024-04-23:
> I'm curious to see if that still happens on the new data APIs. I'll have a look later when I'm back on these branches.

Nope, still happens on `cmc/data_apis_16_non_cacheable_components`.

commit 2641a593 (2024-05-02): Fix mesh flicker (#6157)

Funnily enough, the reason this was flickering on `0.15` was completely
unrelated to the reason why it flickers on `main`.

Long story short: it had to with how explicit client-side splats were
splitting the data across different timestamps and in some cases this
was incorrectly handled.

In the new APIs it's just an issue in the entity_iterator machinery
which lowers everything to range semantics, which leads to problems when
the query expects latest-at semantics.

- Fixes #6072

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/reru [... cut]
closed 2024-05-02 by Wumpf