rerun-io/rerun #6077: Linking to code from docs is dangeround

emilk, 2024-04-23:
Part of
* https://github.com/rerun-io/rerun/issues/1345

We link to files on GitHub quite often in our docs.

For instance, we link to https://github.com/rerun-io/rerun/blob/main/examples/python/log_file/main.py
The problem is that that file is gone.
We should have linked to https://github.com/rerun-io/rerun/blob/0.15.0/examples/python/log_file/main.py instead

So how do we solve this?

* If we use `/blob/main/` the file may change or disappear
* If we use `/blob/latest/` the file may also change or disappear, and the CI may be green just before a release, but the link in the release is wrong
* If we hard-code tags in our links (`/blob/0.15.0/`) our docs will link to old code

jleibs, 2024-04-26:
Just had a user run into errors when trying to run https://www.rerun.io/docs/howto/extend/extend-ui since they copied code from that link into their own project, and ended up using in-flux 0.16 APIs against the 0.15 release.

Wumpf, 2024-05-15:
Everything that is mentioned in an `examples/` README.md should actually be trivial to solve: just do relative paths. `landing` has to understand that ofc then