rerun-io/rerun #6094: `static` components doesn't roundtrip when saved to `.rrd` file

emilk, 2024-04-24:
Repo: load "detect and track objects example", and looks at the streams view:

[image]

Save it to disk, close rerun, and load the saved .rrd. Look again:

[image]

teh-cmc, 2024-04-24:
Oh I don't think we even have a roundtrip test for the save path on the viewer-side :facepalm: (as opposed to the save path on the SDK side).

teh-cmc, 2024-04-30:
> Oh I don't think we even have a roundtrip test for the save path on the viewer-side 🤦 (as opposed to the save path on the SDK side).

Nevermind, we have a pretty extensive test suite for this in fact... what's going on here :thinking:

teh-cmc, 2024-04-30:
I actually cannot repro this. I tried:
- Saving entire timeline, `0.15.1`
- Saving time selection, `0.15.1`
- Saving entire timeline, `main`
- Saving time selection, `main`

emilk, 2024-04-30:
Then let's close this

closed 2024-04-30 by teh-cmc