rerun-io/rerun #6096: Rerun Info messages on launch are spammy and useless

jleibs, 2024-04-24:
None of these are related to an rerun info I care about as a user.
```
$ pixi run rerun
[2024-04-24T11:27:01Z INFO  tracing::span] perform;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=Auth(Some(External), Some([49, 48, 48, 48]))
[2024-04-24T11:27:01Z INFO  tracing::span] read_command;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=NegotiateUnixFD
[2024-04-24T11:27:01Z INFO  tracing::span] read_command;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=Begin
[2024-04-24T11:27:01Z INFO  tracing::span] socket reader;
[2024-04-24T11:27:01Z INFO  tracing::span] perform;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=Auth(Some(External), Some([49, 48, 48, 48]))
[2024-04-24T11:27:01Z INFO  tracing::span] read_command;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=NegotiateUnixFD
[2024-04-24T11:27:01Z INFO  tracing::span] read_command;
[2024-04-24T11:27:01Z INFO  zbus::handshake] write_command; command=Begin
[2024-04-24T11:27:01Z INFO  tracing::span] socket reader;
[2024-04-24T11:27:01Z INFO  zbus::connection] {}; obj_server_task_name="ObjectServer task"
[2024-04-24T11:27:01Z INFO  zbus::object_server] dispatch_message; msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.2"))), path: ObjectPath("/org/a11y/atspi/accessible/root"), iface: InterfaceName(Str(Borrowed("org.freedesktop.DBus.Properties"))), member: MemberName(Str(Borrowed("Set"))), body: Signature("ssv") }
[2024-04-24T11:27:01Z INFO  zbus::object_server] dispatch_method_call; msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.2"))), path: ObjectPath("/org/a11y/atspi/accessible/root"), iface: InterfaceName(Str(Borrowed("org.freedesktop.DBus.Properties"))), member: MemberName(Str(Borrowed("Set"))), body: Signature("ssv") }
[2024-04-24T11:27:01Z INFO  zbus::object_server] dispatch_method_call_try; msg=Msg { type: MethodCall, sender: UniqueName(Str(Borrowed(":1.2"))), path: ObjectPath("/org/a11y/atspi/accessible/root"), iface: InterfaceName(Str(Borrowed("org.freedesktop.DBus.Properties"))), member: MemberName(Str(Borrowed("Set"))), body: Signature("ssv") }
```

emilk, 2024-04-24:
Let's poke the developers and also silence them in `re_log` in the meantime

emilk, 2024-05-06:
Someone on Linux should grab this one

closed 2024-05-08 by jleibs