rerun-io/rerun #6103: API to load and send .rbl file from code

jleibs, 2024-04-24:
To work around situations where blueprint APIs don't yet cover features, users can still manually edit blueprint in the viewer and save them. Unfortunately these blueprints still need to be distributed and manually loaded.

We should add a logging API that opens and existing .rbl file, and sends it to the viewer.

for example:
```
blueprint = rrb.Blueprint.from_file("my_blueprint.rbl")
rr.send_blueprint(blueprint)
```

jleibs, 2024-09-09:
This actually works using the `log_file` API call, but could use some more documentation and an example.

Famok, 2024-11-03:
@jleibs i was looking for this feature. You first proposed a new logging API which I'd very much like to have.
Then you said this already works using log_file. Can you tell me how?

teh-cmc, 2024-11-05:
There's actually a guide for this now:
* https://github.com/rerun-io/rerun/pull/7886

It won't show up on the website until we publish the 0.20 docs though.

closed 2024-11-05 by teh-cmc