rerun-io/rerun #6106: Adjustable depth image range

roym899, 2024-04-24:
It would be nice to make the min and max value for depth images adjustable. Currently the minimum and maximum depth determine the color mapping; however, this can lead to unintuitive jumps in color when a far away value pops in and out of the view. 

This video shows a negative example where the color changes drastically when the sky is not visible.

https://github.com/rerun-io/rerun/assets/9785832/70583d1d-8482-42b2-be0e-38ea93af0896

**Describe the solution you'd like**
- Add an option to adjust the min value and max value for the color map. 
- This setting should be available both for backprojected points and for 2D views. 
- For the backprojection it would also be nice to have an option to exclude points outside of this range. This would improve handling of "non-standard" depth images that use an arbitrary far plane value instead of NaN or 0.

**Additional context**
RViz has this feature ("Normalize Range" can be toggled for depth images and when it's off the "Min Value" and "Max Value" can be set)

Wumpf, 2024-04-25:
Related to
* #1510

makeecat, 2024-08-26:
Do we plan to add this feature in the near future? This small issue can turn down quite a few robotics user from using rerun. Inconsistent depth rendering across frames makes depth data meaningless for analysis.

Wumpf, 2024-09-30:
@makeecat apologies for long wait & lack of reply! It looks like we'll have this finally fixed in the upcoming 0.19 release

closed 2024-10-01 by Wumpf