rerun-io/rerun #6125: Clean up pixi.toml

emilk, 2024-04-26:
Our `pixi.toml` is a bit of a mess. We need to:

* [ ] Define what a lint is
* [ ] Define how we prefix commands
* [ ] Define how we order commands
* [ ] Replace fast_lint.py with pixi?

### Needs
A few high-level commands capturing most of the things a regular users wants to do, including:

* building and running the viewer
* building and activating the python wheel
* running formatting and lints

_after_ that comes the need of the CI which includes:
* Run checks for only one language
* more fine-grained control of the checks

roym899, 2024-05-30:
There's also [this section](https://github.com/rerun-io/rerun/blob/main/BUILD.md#building-an-installable-python-wheel) in BUILD.md mentioning a task to build wheels that does not exist (anymore?).

emilk, 2024-12-11:
A good first step would be to prefix "implementation commands" with `_` to make them "private", e.g.

```toml
# Public command
build-viewer = ["_build_a", "_build_b", …]

# Private/hidden commands
_build_a = […]
_build_b = […]
```

I don't know if this is official sanctioned pixi syntax, but it could be