rerun-io/rerun #6138: Entity path with combined Transform3D + Pinhole as origin makes it impossible to visualize any parents

roym899, 2024-04-26:
**Describe the bug**
When combining a 3D transform and a pinhole transform on the same entity path and setting this path as the origin of a 3D view, no parents / siblings can be visualized in such a view anymore.

This makes it impossible to let the view follow a moving camera, for example.

**To Reproduce**
```python
import rerun as rr
import numpy as np

rr.init("merged pinhole + transform3d", spawn=True)

rr.log("", rr.Points3D(np.random.rand(100, 3), colors=(255,0,0)))

rr.log("camera", rr.Transform3D(translation=(0, 5, 0)))
rr.log("camera", rr.Pinhole(resolution=(500, 500), principal_point=(250, 250), focal_length=(500, 500)))
```
Setting camera as the origin, the points cannot be visualized anymore.

**Expected behavior**
I'd expect the origin to be at the origin of the frustum, but other entities still to be addable.

**Rerun version**
```
rerun_py 0.15.0 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.15.0 bc8bcd0, built 2024-04-09T14:34:04Z
```

Wumpf, 2024-04-27:
siblings & parents should work fine if you add them to the `contents` of the space view - default contents are everything under the origin, but you can also set it to e.g. `/**` and then it should work fine

roym899, 2024-04-27:
Yes, I don't think it works.

closed 2025-10-06 by Wumpf