rerun-io/rerun #6140: Add a way to disable the welcome screen.

jleibs, 2024-04-26:
While the welcome screen makes sense for users that are primary users of Rerun, it can be confusing to "secondary users", when Rerun is used as part of a visualizer spawned by another application.

emilk, 2024-05-06:
You want to set this as an `init` argument, or as a blueprint option, or what?

jleibs, 2024-05-06:
Maybe even just an argument to spawn?

Doing it from blueprint would be nice too, but introduces some weird edge-cases when multiple blueprints conflict on what to do. 

Would also like to be able to do it from Gradio, so a url-param would be nice there as well.

Wumpf, 2024-05-15:
@jleibs now that https://github.com/rerun-io/rerun/pull/6262 is in, what is still missing? (that pr says it was partially addressed)

pd-nisse, 2024-05-24:
@jleibs thanks for `spawn` argument, that works well for us. I noticed it isn't supported in `serve`, though, which is what we use in headless mode. Any chance to expand that argument in the next version?