rerun-io/rerun #6143: Clicking on suggestion does not select it

roym899, 2024-04-27:
**Describe the bug**
Selecting a space origin from the suggestions does not select it when clicking. Only the list of suggestions is closed.

Pressing enter while one suggestion is highlighted works.

**Video**

https://github.com/rerun-io/rerun/assets/9785832/4b29e70f-8514-4dd2-9213-4b2dd4e3cbe8

**Desktop (please complete the following information):**
 - Ubuntu 22.04

**Rerun version**
> rerun_py 0.15.0 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.15.0 bc8bcd0, built 2024-04-09T14:34:04Z

Wumpf, 2024-04-27:
gave this a `user-request` just to priotize it - this just looks so _insanely_ annoying Oo

emilk, 2024-05-02:
If you click slowly you'll notice that the popup closes on the press, not the release (and clicks are registered on release).
So the popup is closing before the click is registered.
The reason is that the text edit looses focus when the user presses somewhere else, and that switches us to `SpaceOriginEditState::NotEditing`, and then the code path for showing the popup ui isn't run.

closed 2024-05-02 by emilk