rerun-io/rerun #6185: Profilng breakdown of camera -> display latency

jleibs, 2024-05-01:
I did a very adhoc test to measure closed-loop latency using a camera to capture the screen, and presenting the captured image along side the frame  comparing to cv.imshow as a reference.

Can be reproduced using the following gist:
https://gist.github.com/jleibs/08c65d6195ea5dad79e8974201258f62

My tests were on with a 1600x900 camera running at 60fps 
We can see a total round-trip phase-lag for Rerun of 9-10 frames (160ms), and for `cv.imshow` of about 8 frames.

1-frame discprency between Rerun and Imshow
[image]

2-frame disprepency between Rerun and Imshow
[image]

This 1-2 frame lag seems inline with the Rerun-measured latency of 13ms + the app/render latency of 4ms + the additional random phase offset between the display clock and the camera clock.

jleibs, 2024-05-01:
It would be good to run a similar comparison using a Rust or C++ app with a more efficient camera -> app pipeline.

emilk, 2024-05-02:
The default value of `RERUN_FLUSH_TICK_SECS` is also 8ms: https://www.rerun.io/docs/reference/sdk-micro-batching

jleibs, 2024-05-02:
> The default value of `RERUN_FLUSH_TICK_SECS` is also 8ms

Good point. I don't think that should be coming into play here since `RERUN_FLUSH_NUM_BYTES` is 1MB and each image is over 4MB