rerun-io/rerun #6193: Unable to start rerun viewer on clean Ubuntu 22.04

changh95, 2024-05-02:
**Describe the bug**

Rerun viewer crashes on starting.

**To Reproduce**

1. Clean install of Ubuntu 22.04. Hardware-wise, there's an Intel CPU and NVIDIA RTX 3080 GPU.
2. Installed rerun-sdk via PyPI.
3. Started rerun viewer with 'rerun' on bash terminal.

**Expected behavior**

Rerun viewer is supposed to open.

**Screenshots**

```
~ ❯❯❯ pip install rerun-sdk
Defaulting to user installation because normal site-packages is not writeable
Collecting rerun-sdk
  Using cached rerun_sdk-0.15.1-cp38-abi3-manylinux_2_31_x86_64.whl.metadata (5.2 kB)
Requirement already satisfied: attrs>=23.1.0 in ./.local/lib/python3.10/site-packages (from rerun-sdk) (23.2.0)
Requirement already satisfied: numpy<2,>=1.23 in ./.local/lib/python3.10/site-packages (from rerun-sdk) (1.26.4)
Requirement already satisfied: pillow in ./.local/lib/python3.10/site-packages (from rerun-sdk) (10.3.0)
Requirement already satisfied: pyarrow>=14.0.2 in ./.local/lib/python3.10/site-packages (from rerun-sdk) (16.0.0)
Requirement already satisfied: typing-extensions>=4.5 in ./.local/lib/python3.10/site-packages (from rerun-sdk) (4.11.0)
Using cached rerun_sdk-0.15.1-cp38-abi3-manylinux_2_31_x86_64.whl (27.2 MB)
Installing collected packages: rerun-sdk
Successfully installed rerun-sdk-0.15.1
~ ❯❯❯ rerun
[2024-05-02T12:09:24Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: IntegratedGpu, name: "Intel(R) Graphics (RPL-S)", driver: "Intel open-source Mesa driver", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2", vendor: 0x8086, device: 0xA780}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 15.0.7, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2 (LLVM 15.0.7)", vendor: 0x10005}, {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 3080", driver: "NVIDIA", driver_info: "535.171.04", vendor: 0x10DE, device: 0x2206}

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   1: re_crash_handler::install_signal_handler::signal_handler
   2: <unknown>
   3: _dl_close
             at dl-close.c:788:3
   4: __GI__dl_catch_exception
             at dl-error-skeleton.c:208:8
   5: __GI__dl_catch_error
             at dl-error-skeleton.c:227:19
   6: _dlerror_run
             at dlerror.c:138:17
   7: __dlclose
             at dlclose.c:31:10
   8: ProducerCleanup
   9: ProducerInit
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: wgpu_hal::vulkan::instance::<impl wgpu_hal::Surface<wgpu_hal::vulkan::Api> for wgpu_hal::vulkan::Surface>::configure
  16: wgpu_core::device::global::<impl wgpu_core::global::Global<G>>::surface_configure
  17: <wgpu::backend::wgpu_core::ContextWgpuCore as wgpu::context::Context>::surface_configure

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

[1]    185204 segmentation fault (core dumped)  rerun
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Intel CPU, NVIDIA RTX 3080 (Driver 535.171.04)

**Rerun version**
0.15.1

**Additional context**

Wumpf, 2024-05-02:
from the looks of the log its crashing inside one of your graphics drivers trying to create a vulkan surface. This could be a wayland issue as described here https://github.com/gfx-rs/wgpu/issues/5487

Can you try running with `WAYLAND_DISPLAY=""`?

jleibs, 2024-05-02:
Also, helpful for these kinds of crash reports, set `RUST_LOG=debug`, which should help us see which of the adapters is being chosen.

changh95, 2024-05-03:
> from the looks of the log its crashing inside one of your graphics drivers trying to create a vulkan surface. This could be a wayland issue as described here [gfx-rs/wgpu#5487](https://github.com/gfx-rs/wgpu/issues/5487)
> 
> * (possibly related to [Intermittent Device Lost crash in wgpu on wayland  #1250](https://github.com/rerun-io/rerun/issues/1250))
> 
> Can you try running with `WAYLAND_DISPLAY=""`?

This was the solution!
I forgot I'm running my system on Wayland.

changh95, 2024-05-03:
> Also, helpful for these kinds of crash reports, set `RUST_LOG=debug`, which should help us see which of the adapters is being chosen.

Despite setting `RUST_LOG=debug`, I'm getting the same log output.

```
~ ❯❯❯ RUST_LOG=debug                                                                                                                             ✘ 139 
~ ❯❯❯ rerun
[2024-05-03T02:38:10Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: IntegratedGpu, name: "Intel(R) Graphics (RPL-S)", driver: "Intel open-source Mesa driver", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2", vendor: 0x8086, device: 0xA780}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 15.0.7, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2 (LLVM 15.0.7)", vendor: 0x10005}, {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 3080", driver: "NVIDIA", driver_info: "535.171.04", vendor: 0x10DE, device: 0x2206}

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   1: re_crash_handler::install_signal_handler::signal_handler
   2: <unknown>
   3: _dl_close
             at dl-close.c:788:3
   4: __GI__dl_catch_exception
             at dl-error-skeleton.c:208:8
   5: __GI__dl_catch_error
             at dl-error-skeleton.c:227:19
   6: _dlerror_run
             at dlerror.c:138:17
   7: __dlclose
             at dlclose.c:31:10
   8: ProducerCleanup
   9: ProducerInit
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: wgpu_hal::vulkan::instance::<impl wgpu_hal::Surface<wgpu_hal::vulkan::Api> for wgpu_hal::vulkan::Surface>::configure
  16: wgpu_core::device::global::<impl wgpu_core::global::Global<G>>::surface_configure
  17: <wgpu::backend::wgpu_core::ContextWgpuCore as wgpu::context::Context>::surface_configure

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

[1]    212872 segmentation fault (core dumped)  rerun
```

Wumpf, 2024-05-03:
Writing `RUST_LOG=debug` will in almost all shells only set the variable for the current command. So you need to do `RUST_LOG=debug rerun`

changh95, 2024-05-03:
```
~ ❯❯❯ RUST_LOG=debug rerun                                               ✘ 139 
[2024-05-03T08:44:42Z DEBUG rerun::run] Detected 32 cores. Using 30 compute threads.
[2024-05-03T08:44:42Z DEBUG re_memory::memory_limit] Setting memory limit to 23.3 GiB, which is 75% of total available memory (31.1 GiB).
[2024-05-03T08:44:42Z DEBUG re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-05-03T08:44:42Z DEBUG eframe] Using the wgpu renderer
[2024-05-03T08:44:42Z DEBUG sctk] Bound new global [28] wl_output v3
[2024-05-03T08:44:42Z DEBUG sctk] Bound new global [5] zxdg_output_manager_v1 v3
[2024-05-03T08:44:42Z DEBUG sctk] Bound new global [16] wl_seat v5
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Argb8888
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xrgb8888
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Abgr8888
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xbgr8888
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Rgb565
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Argb2101010
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xrgb2101010
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Abgr2101010
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xbgr2101010
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Argb16161616f
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xrgb16161616f
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Abgr16161616f
[2024-05-03T08:44:42Z DEBUG sctk] supported wl_shm format Xbgr16161616f
[2024-05-03T08:44:42Z DEBUG eframe::native::wgpu_integration] Event::Resumed
[2024-05-03T08:44:42Z DEBUG eframe::native::file_storage] Loading app state from "/home/stradvision/.local/share/rerun/app.ron"…
[2024-05-03T08:44:42Z DEBUG egui_winit] Failed to set window size
[2024-05-03T08:44:42Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: IntegratedGpu, name: "Intel(R) Graphics (RPL-S)", driver: "Intel open-source Mesa driver", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2", vendor: 0x8086, device: 0xA780}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 15.0.7, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2 (LLVM 15.0.7)", vendor: 0x10005}, {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 3080", driver: "NVIDIA", driver_info: "535.171.04", vendor: 0x10DE, device: 0x2206}
[2024-05-03T08:44:42Z DEBUG egui_wgpu] Picked wgpu adapter: backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 3080", driver: "NVIDIA", driver_info: "535.171.04", vendor: 0x10DE, device: 0x2206

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   1: re_crash_handler::install_signal_handler::signal_handler
   2: <unknown>
   3: _dl_close
             at dl-close.c:788:3
   4: __GI__dl_catch_exception
             at dl-error-skeleton.c:208:8
   5: __GI__dl_catch_error
             at dl-error-skeleton.c:227:19
   6: _dlerror_run
             at dlerror.c:138:17
   7: __dlclose
             at dlclose.c:31:10
   8: ProducerCleanup
   9: ProducerInit
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: wgpu_hal::vulkan::instance::<impl wgpu_hal::Surface<wgpu_hal::vulkan::Api> for wgpu_hal::vulkan::Surface>::configure
  16: wgpu_core::device::global::<impl wgpu_core::global::Global<G>>::surface_configure
  17: <wgpu::backend::wgpu_core::ContextWgpuCore as wgpu::context::Context>::surface_configure

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

[1]    61059 segmentation fault (core dumped)  RUST_LOG=debug rerun
```