rerun-io/rerun #6254: Measurement Tool for 3D Scenes

tsugg, 2024-05-07:
**Is your feature request related to a problem? Please describe.**

I would like to use Rerun as an inspection tool where I can collect Lidar and image data at a facility, and remote inspect at a later date. I would like the ability to measure the distance between 2 points in a 3D Scene, since that is an important task in my inspection use case.

**Describe the solution you'd like**

When holding Ctrl + Left Click on 2 points, a line will display between the points and either a small window will display with the distance, or it will populate in the selection panel. I think a line segment between points would be valuable regardless of the solution to help someone visualize the distance they are searching for.

**Describe alternatives you've considered**

Point information is visible when Ctrl + Left Click on  any point. A user could use a separate app or a calculator to get the measurement distance. CloudCompare or another point cloud viewing tool could be used. However, this makes the inspection process longer and more cumbersome. 

**Additional context**

Example scene with the desired feature ...

[image]

[image]

vinnnyr, 2024-07-03:
I also think this in the 2D case would be very helpful! In fact for my application I would use it more in the 2D case. 

Even if we display current mouse coordinates, that could help enable user to do measurements between key areas themselves.

dalux-mv, 2026-08-25:
I think this would be a nice addition to the tool as well.

It would be nice to have point to point measurement tool to calculate distance between two objects in Viewport.