rerun-io/rerun #6263: UI Design for a "No Data" page when the Welcome screen has been disabled

jleibs, 2024-05-07:
- https://github.com/rerun-io/rerun/pull/6262 made it possible to disable the welcome screen.

This was motivated by use-cases where Rerun is being embedded in another product context. Indirect users (those that have not logged the data and know nothing about Rerun) aren't sure what to do with the Welcome screen.

However, the question is what do we show in this case when all the recordings have been closed.

Ideally we would have something that looked nicer than this while fitting within the spirit of 
[image]

abey79, 2024-05-08:
@nikolausWest We could do with your input here. I see two possible avenue: either fully minimalist, or we inject a little bit of context about the viewer is about using a light version of the welcome screen.

#### Minimalist

Just to pieces of copy, in weak still:
- placeholder where the recording would show up
- placeholder in the middle of the viewport

[image]

#### Light welcome screen

- Same "no recording" placeholder
- Similar heading as the current welcome screen, though the copy might have to be adjusted to be not too surprising for "data-consumer"-type users

[image]

abey79, 2024-05-10:
Let's start with this version, with "No recording**s** loaded" as copy in the recording list.

[image]

PR #6287 (2024-05-10): Improve the default UI when the welcome screen is hidden
### What

Follow-up of #6262 to improve the default UI when the welcome screen is disabled and no recordings are loaded.

- Fixes #6263

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using examples from latest `main` build: [rerun.io/viewer](https://reru [... cut]
commit bc689400 (2024-05-13): Improve the default UI when the welcome screen is hidden (#6287)

### What

Follow-up of #6262 to improve the default UI when the welcome screen is
disabled and no recordings are loaded.

- Fixes #6263

<img width="1522" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/ce7355e3-2aad-4171-9187-cda18df8aeaa">

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
 [... cut]