rerun-io/rerun #6265: Improvements to BarChart

mvarble, 2024-05-07:
Currently, the bar chart only tracks the heights of the bins and not the "edges" of each bin. An unfortunate downside of this is that when I have a distribution of scalar data that I would like to view in a histogram, I have no idea where the mean is. Consider an animation like below, where the _number_ of bins stays constant, but the mean moves.

[image]

I believe animations such as this could be achieved by extending the `BarChart` component to have `edges` and the `BarChart` blueprint to have view bounds.

Wumpf, 2024-05-08:
Very related (aaaalmost duplicate) to:
* #4764

mvarble, 2024-05-08:
> Very related (aaaalmost duplicate) to:
> 
>     * [Allow to specify boundaries for BarChart #4764](https://github.com/rerun-io/rerun/issues/4764)

Gah, I am sorry for spam. I must have searched _histogram_ or some other related term that didn't cause a match.

Wumpf, 2024-05-09:
no worries, I'm keeping this open since it's actually a bit of a different aspect

closed 2025-12-16 by Wumpf