rerun-io/rerun #6275: (Optionally) Prevent projections behind the camera plane

genemerewether, 2024-05-09:
3D objects also project into cameras that are pointed the opposite direction, giving a negative depth value in the camera image view.

Could we prevent that (check that projected depth is positive)? It's confusing when you see a point project to cameras that are pointed the opposite way.

Wumpf, 2024-05-13:
marking this as a bug since this was not the intended behavior.
To clarify: You're seeing 3d objects in the Spatial2DView that are behind the projecting pinhole camera? As far as you know, does this apply to specific objects?

genemerewether, 2024-05-13:
Not static objects - I’m not sure the right word. Selection (hover mouse
over 3D window) in 3D transiently highlights an object, then shows the
depth of that object in camera 2D views. This happens even for objects
behind cameras, at negative depth.

Sorry, I can’t screenshot our actual app and I don’t have a repro, but I
can sketch it if that helps

Wumpf, 2024-05-13:
ahhh makes sense, it's the ui interactions that are shown. I thought we had a bug for 3d objects on 2d views. Makes sense!

Wumpf, 2024-10-22:
let's hide negative values always. making this a setting is more confusing than helpful imho