rerun-io/rerun #6276: TimeBoundary types should also support floating point seconds

jleibs, 2024-05-09:
We expose these for our set_time APIs, so it's surprising floats cast to int in an unexpected way.

Wumpf, 2024-05-13:
nothing to add to the first point and I think it's quite important to improve this as suggested.

> API doesn't make it clear you can't specify a time range more than one

it is documented that it's a first value wins situation `/// If a timeline is listed twice, the first entry will be used.`.
We could look into making this also take a dictionary but that gets tricky to map out to the existing data structures. I'm not convinced yet this is that much of a problem to justify a lot more time spent on it.

jleibs, 2024-05-13:
> it is documented that it's a first value wins situation

Not in the doc visible in the editor:
[image]

This other thing I was thinking we could do would just be scan them and print a warning.

commit 15511fa6 (2024-05-14): Better handle redundant time ranges (#6306)

### What

* Fixes #6276
* Better doc visibility
* Warn if a timeline is specified twice

Collateral:
* generate `__hash__` methods for trivial python classes
* fix issue with `pixi run py-test`

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using examples from latest `main` bu [... cut]
closed 2024-05-14 by Wumpf