rerun-io/rerun #6286: running with serve() of a really short script can make the web result close before data is rendering

YoniChechik, 2024-05-10:
**Describe the bug**

running with serve() of a really short script can make the web result close before data is rendering, and getting the welcome screen instead.

**To Reproduce**
Steps to reproduce the behavior:
```python
import time

import numpy as np
import rerun as rr

rr.init("data")

rr.serve()
# time.sleep(1) #ENABLE THIS AS A CURRENT WORKAROUND
positions = np.zeros((10, 3))
positions[:, 0] = np.linspace(-10, 10, 10)

colors = np.zeros((10, 3), dtype=np.uint8)
colors[:, 0] = np.linspace(0, 255, 10)

rr.log("my_points", rr.Points3D(positions, colors=colors, radii=0.5))
```

from this we get a welcome screen SOMETIMES instead of the points.

**Expected behavior**
Always get the results webview

**Desktop (please complete the following information):**
 - OS: windows 11, conda.

**Rerun version**

rerun_py 0.15.1 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-pc-windows-msvc release-0.15.1 7dedf88, built 2024-04-11T14:56:57Z

emilk, 2024-05-14:
The problem here is that the serve shuts down when the script shuts down.

We should improve the docs of `serve()` to explain this.

We should also consider other fixes:
* Log a warning if the server shuts down to quickly
* Add a `block_on_exit` argument or simimlar

Note that this problem affects all logging SDKs

emilk, 2024-05-15:
I'm digging into this a little bit more. 

If I run this simple script:

```py
rr.init("rerun_example_data")
rr.serve()
rr.log("text", rr.TextDocument("Hello World!"))
```

My browser does start, downloads the viewer Wasm, and connects over web-socket - but the "Hello World!" string is never sent over!

emilk, 2024-05-15:
The problem seems to be that `WebViewerSink` doesn't implement `flush_blocking`, so the last log message(s) may never arrive to the client.

closed 2024-05-15 by emilk