rerun-io/rerun #6330: Further check and refactor unwrap() usage

Artxiom, 2024-05-15:
@Wumpf 
Follow up for #3408.
After the initial refactor there are still some crates left which use unwrap:

- [ ] re_data_store
- [ ] re_dev_tools
- [ ] re_entity_db
- [ ] re_format_arrow
- [ ] re_log_types
- [ ] re_query
- [ ] re_renderer
- [ ] re_renderer_examples/*
- [ ] re_sdk
- [ ] re_space_view_spatial
- [ ] re_space_view_tensor
- [ ] re_space_view_time_series
- [ ] re_time_panel
- [ ] re_types_builder
- [ ] re_types_core
- [ ] re_viewer
- [ ] re_viewport
- [ ] re_ws_comms
- [ ] snippets

They are all marked with:
```
// TODO(#3408): remove unwrap()
#![allow(clippy::unwrap_used)]
```

ilyaaay, 2024-08-01:
Hello. I want to take this issue (I started with re-renderer)

Artxiom, 2024-08-03:
@ilyaaay I realized it's not a huge amount of changes yet so I pushed my branch [here](https://github.com/Artxiom/rerun/tree/6330-refactor-unwrap-re_renderer).

Feel free to pick it up - I will be super busy probably until beginning of next year.
Just please `@mention` me somewhere when you do so, so I don't pick it up accidentally again.

riccardocinti, 2024-09-14:
Hi everyone. I'd like to contribute to this issue and I started working on the top/re_sdk and the store/re_sdk_comms crates.

emilk, 2025-09-15:
Closed in favor of the original issue:
* https://github.com/rerun-io/rerun/issues/3408

closed 2025-09-15 by emilk