rerun-io/rerun #6357: Image width/height/channel method helper seems broken

teh-cmc, 2024-05-16:
```rust
    dbg!((
        &tensor.shape,
        tensor.shape_short(),
        [height, width, channel]
    ));
```
outputs:
```
[crates/re_viewer_context/src/gpu_bridge/tensor_to_gpu.rs:556:5] (&tensor.shape, tensor.shape_short(), [height, width, channel]) = (
    [
        width=1,
        height=1,
        depth=4,
    ],
    [
        depth=4,
    ],
    [
        4,
        1,
        1,
    ],
)
```
which then results in:
```
[2024-05-16T12:58:07Z WARN  re_space_view_spatial::mesh_cache] Failed to load mesh "/world/65v1/comps/sprite": Only 3 and 4 channel u8 tensor data is supported currently for mesh textures, got 1.
```

Looks like a logic issue in:
```rust
    pub fn image_height_width_channels(&self) -> Option<[u64; 3]> {
        let shape_short = self.shape_short();
        match &self.buffer {
            // In the case of NV12, return the shape of the RGB image, not the tensor size.
            TensorBuffer::Nv12(_) => {
                // NV12 encodes a color image in 1.5 "channels" -> 1 luma (per pixel) + (1U+1V) / 4 pixels.
                match shape_short {
                    [h, w] => Some([h.size * 2 / 3, w.size, 3]),
                    _ => None,
                }
            }
            // In the case of YUY2, return the shape of the RGB image, not the tensor size.
            TensorBuffer::Yuy2(_) => {
                // YUY2 encodes a color image in 2 "channels" -> 1 luma (per pixel) + (1U + 1V) (per 2 pixels).
                match shape_short {
                    [h, w] => Some([h.size, w.size / 2, 3]),
                    _ => None,
                }
            }
            TensorBuffer::Jpeg(_)
            | TensorBuffer::U8(_)
            | TensorBuffer::U16(_)
            | TensorBuffer::U32(_)
            | TensorBuffer::U64(_)
            | TensorBuffer::I8(_)
            | TensorBuffer::I16(_)
            | TensorBuffer::I32(_)
            | TensorBuffer::I64(_)
            | TensorBuffer::F16(_)
            | TensorBuffer::F32(_)
            | TensorBuffer::F64(_) => {
                match shape_short.len() {
                    1 => {
                        // Special case: Nx1(x1x1x …) tensors are treated as Nx1 gray images.
                        // Special case: Nx1(x1x1x …) tensors are treated as Nx1 gray images.
                        if self.shape.len() >= 2 {
                            Some([shape_short[0].size, 1, 1])
                        } else {
                            None
                        }
                    }
                    2 => Some([shape_short[0].size, shape_short[1].size, 1]),
                    3 => {
                        let channels = shape_short[2].size;
                        if matches!(channels, 1 | 3 | 4) {
                            // mono, rgb, rgba
                            Some([shape_short[0].size, shape_short[1].size, channels])
                        } else {
                            None
                        }
                    }
                    _ => None,
                }
            }
        }
    }
```

emilk, 2024-05-16:
The problem is that `shape_short` ignores _leading_ dim=1 since https://github.com/rerun-io/rerun/pull/5579

emilk, 2024-05-16:
The usage of `shape_short` in `Nv12` and `Yuy2` is also sus

PR #6360 (2024-05-16): Fix width/height/channel extraction logic
- Fixes #6357

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using examples from latest `main` build: [rerun.io/viewer](https://rerun.io/viewer/pr/6360?manifest_url=https://app.rerun.io/version/main/examples_manifest.json)
  * Using full set of examples from `nightly` build: [r [... cut]
commit 838eab1b (2024-05-16): Fix width/height/channel extraction logic (#6360)

- Fixes #6357

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using examples from latest `main` build:
[rerun.io/viewer](https://rerun.io/viewer/pr/6360?manifest_url=https://app.rerun.io/version/main/examples_manifest.json)
* Using full set of examples from `nightly` buil [... cut]
commit 404dcb24 (2024-05-16): Fix width/height/channel extraction logic (#6360)

- Fixes #6357

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using examples from latest `main` build:
[rerun.io/viewer](https://rerun.io/viewer/pr/6360?manifest_url=https://app.rerun.io/version/main/examples_manifest.json)
* Using full set of examples from `nightly` buil [... cut]
closed 2024-05-16 by emilk