rerun-io/rerun #6358: Selecting an instance containing a batch of keypoint_id + class_id spams a warning: "not a mono-batch"

jleibs, 2024-05-16:
Easy repro using the face_tracking example.

```
python -m face_tracking
```
then select any keypoint.

[image]

Text of the warning:
```
Processing frames: 691it [00:23, 27.51it/s][2024-05-16T13:16:52Z WARN  re_query::latest_at::helpers] Couldn't deserialize /reconstruction/faces/0:rerun.components.ClassId @ TimeInt(690)#17CFFA806598864D24D999916925D29D: not a mono-batch (length: 478)
```

### Root cause

The cause of the problem comes from:
https://github.com/rerun-io/rerun/blob/692b6b6acb185f96d90e0e5a7bf392d6a802d0a3/crates/re_data_ui/src/annotation_context.rs#L97-L99

We are using `latest_at_component` which depends on a mono-component, but the keypoint is an instance of a batch.

To fix this properly, we need to track the instance all the way through EntityDataUi and use it to look up the correct class-id.

### Affects
0.15, 0.16

jleibs, 2024-05-16:
In: https://github.com/rerun-io/rerun/pull/6359 we made this incrementally better by getting rid of the log spam

emilk, 2024-05-16:
Looks like this issue is finally producing real problems:
* https://github.com/rerun-io/rerun/issues/3168

emilk, 2024-05-16:
In fact, I'm closing this as a duplicate of https://github.com/rerun-io/rerun/issues/3168

closed 2024-05-16 by emilk