rerun-io/rerun #6367: Please make memory-stats optional based on compile-time switch because it doesn't yet support BSDs

yurivict, 2024-05-17:
https://github.com/Arc-blroth/memory-stats/issues/9

Memory stats shouldn't be an absolutely required feature. It's better to have rerun buildable without memory stats than not buildable at all.

Wumpf, 2024-05-17:
Should be easy enough to accommodate for this, but I wouldn't be surprised if there's more issues when building for FreeBSD

emilk, 2024-05-17:
I think this should be fixed in `memory-stats` instead - it should ideally compile on every platform, but return `None` on unsupported platforms.

emilk, 2024-05-17:
…but I guess that crate isn't getting much love these days: https://github.com/Arc-blroth/memory-stats/pull/8

yurivict, 2024-05-19:
IMO, it is easier to add a compile-time switch in rerun to unblock this issue.
Then, if memory-status would change later - this can be adopted, when they change.

emilk, 2024-05-20:
For sure - just excluding `memory-stats` on BSD sounds like a good idea. Do you wanna make a PR?

emilk, 2024-05-20:
…or better yet, maybve `memory-stats` can be replaced by https://crates.io/crates/sys-info

dakshbtc, 2024-05-26:
Can i pick this . please

emilk, 2024-05-27:
Since https://github.com/Arc-blroth/memory-stats/pull/8 is merged we're only waiting for a new release of `memory-stats` to have this solved

emilk, 2024-05-27:
@dakshbtc do you want to replace our use of `memory-stats` with `sys-info`?

dakshbtc, 2024-06-01:
@emilk yes i can do that.

dakshbtc, 2024-06-01:
it is just being used in ./rerun/crates/re_memory/src/memory_use.rs file currently. Right?