rerun-io/rerun #6375: Point clouds are not correctly rendered on mobile browsers.

jlecoeur, 2024-05-17:
**Describe the bug**
3D Point clouds are not correctly rendered by the viewer on mobile browsers. 
Related to https://github.com/rerun-io/rerun/issues/1672.

**To Reproduce**
Load example https://www.rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.16.0%2Fexamples%2Fnuscenes_dataset.rrd on a mobile browser.

**Expected behavior**
Same as native viewer and desktop browser.

**Screenshots**
[image]

If we increase the UI point size to a very big value, the spheres do show up, but appear to be missing faces:

[image]

**Desktop (please complete the following information):**
 - OS: Android (11)
 - Browser: tested with Firefox (125.3.0) and Chrome (124.0)

**Rerun version**
0.16

Wumpf, 2024-05-17:
Likely the same issue as 
* https://github.com/rerun-io/rerun/issues/5227

Wumpf, 2024-05-17:
@jlecoeur  do you see the same issue in this demo https://www.rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.16.0%2Fexamples%2Fstructure_from_motion.rrd? I have the strong suspicion that this affects only point clouds that are generated from depth images

jlecoeur, 2024-05-17:
Yes there is the same problem on the colmap example.

I also see the issue with a custom script that logs a series of `Points3D`.

jlecoeur, 2024-05-17:
Here is more information for you, the issue does not affect all devices:
- 👎 Xiaomi Redmi note 9S, Android 11, Firefox 125.3.0
- 👎 Xiaomi 11 Lite 5G, Android 11, Chrome 125.0.6422.53
- 👎 Galaxy Tab S5e, Android 11, Firefox 121.1.0, Chrome 124.0.6367.179, Brave 1.65.132 (Chromium 124.0.6367.202)
- 🆗 Galaxy Tab A 2016, Android 8.1, Brave 1.65.132 (Chromium 124.0.6367.202)
- 🆗 Asus Zenfone 9, Android 14, Firefox 126.0, Brave 1.65.133 (Chromium 124.0.6367.208)