rerun-io/rerun #6379: Version Comparison Macros for C++

jleibs, 2024-05-17:
It would be beneficial to provide a method to check versions using macro/preprocessors for cases where the code must support multiple versions.

Although there is the [RERUN_SDK_HEADER_VERSION](https://www.internalfb.com/code/fbsource/[06ebd76cc519]/third-party/rerun_cpp_sdk/src/rerun/c/rerun.h?lines=271) preprocessor available, its value is a string (e.g., "0.15.0"), making comparison difficult. It would be more practical to change it to a comparable value like 001500 or to provide separate version numbers for major, minor, and patches that can be compared using a utility macro such as:
```
#define RERUN_SDK_HEADER_VERSION_MAJOR 0
#define RERUN_SDK_HEADER_VERSION_MINOR 15
#define RERUN_SDK_HEADER_VERSION_PATCH 0
#define RERUN_SDK_HEADER_VERSION_GT(major, minor, patch) ...
```

emilk, 2024-06-11:
We have a similar need in Python

jeongseok-meta, 2024-07-09:
I wonder if this issue is addressed in the 0.17.0 release. Unfortunately, the 0.17.0 version breaks the build with code for 0.16.0, for example:

https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=973987&view=logs&j=58ac6aab-c4bc-5de2-2894-98e408cc8ec9&t=933f325c-924e-533d-4d95-e93b5843ce8b&l=1404

jleibs, 2024-07-09:
> I wonder if this issue is addressed in the 0.17.0 release. Unfortunately, the 0.17.0 version breaks the build with code for 0.16.0

Sadly we still haven't managed to prioritize this yet.

Out of curiosity does `momentum` need to maintain single-source builds against both 0.16 and 0.17 simultaneously?

jeongseok-meta, 2024-07-09:
Got it. Maybe we need to wait for the next releases then.

Regarding supporting multiple versions, we would like to avoid that if possible, but there is a case where the internal rerun SDK version is not yet updated while momentum's [build recipe at conda-forge](https://github.com/conda-forge/momentum-feedstock/blob/main/recipe/meta.yaml#L45) needs to support the latest version as well.

This can be worked around by restricting the rerun SDK version for the momentum recipe. One downside would be that it will also restrict the downstream packages of momentum to use specific versions of the rerun SDK and viewer.

emilk, 2024-08-07:
I tried approaching this similar to the current code with:

```C
#define RERUN_SDK_HEADER_VERSION "@RERUN_VERSION@"

#define RERUN_SDK_HEADER_VERSION_MAJOR @RERUN_VERSION_MAJOR@
#define RERUN_SDK_HEADER_VERSION_MINOR @RERUN_VERSION_MINOR@
#define RERUN_SDK_HEADER_VERSION_PATCH @RERUN_VERSION_PATCH@
```

With a this in `CMakeLists.txt`:

```cmake
string(REGEX MATCH "\nversion = \"([0-9]+)\.([0-9]+).([0-9]+)[a-z0-9\+-]+\"\n" _ ${CARGO_TOML})
set(RERUN_VERSION_MAJOR ${CMAKE_MATCH_1})
set(RERUN_VERSION_MINOR ${CMAKE_MATCH_2})
set(RERUN_VERSION_PATCH ${CMAKE_MATCH_3})
```

But apparently the magical `@FOO@` syntax only works for strings, so we need to reach for a different tool than CMake.

emilk, 2024-08-08:
We can follow the lead of
* https://github.com/rerun-io/rerun/pull/7104

closed 2024-08-12 by emilk