rerun-io/rerun #6388: Less arrow unions

emilk, 2024-05-20:
Arrow unions has downsides:
* slow serialization/deserialization (no zero-copy)
* hard to codegen (especially for Python)
* complex for users that want the raw arrow data

### Transform
We can split the `Transform` union-component into an archetype:

```
archetype Transform {
    // if set, all other fields are ignored
    mat4: Option<Mat4>,

    translation: Option<Translation3>,

    // scale, rotation, shear. If set, all other fields are ignored.
    mat3: Option<Mat3>,

    rotation: Option<Rotation3D>,

    // if set, the uniform `Scale` is ignored
    scale3: Option<Scale3D>,

    scale: Option<Scale>,
}
```

jleibs, 2024-05-27:
Some additional notes on the above:

### Why should Images use an untyped buffer + pixel format while tensors use a typed buffer?

While at first glance this proposal might seem to introduce an inconsistency, in practice it serves to highlight the fundamental differences between these two approaches to data representation.

Images are a way of describing a (possibly multi-channel) pixel value over a 2D image plane.

Images are almost always specifically grounded in data received from sensors or sent to displays. This usage, as it relates to special-built hardware, has given rise to pragmatic ways of describing these pixel values more efficiently for purposes of implementation. It is not uncommon for pixel encodings to pack data in ways that simply don't align with a uniform-shape tensor representation. See, chroma subsampling, bayer patterns, etc. It is also quite common to consider an approximate or interpolated pixel value as the data is inherently 2d-spatial.

As such a raw buffer + image encoding really is the most authentic representation we can achieve.  For many low-level image libraries or sensor drivers we should be able to directly map this structure to an API that lets us access or load the raw image buffer + some metadata.

On the other hand, Tensors are much more generally mapped to multi-dimensional arrays. They are often used in pure data and computational contexts that have nothing to do with images. Due to the wildly varied applications, the patterns of tensor compression (beyond things like run-length-encoding, or sparse / dense representation) are much more varied and domain specific. This means there simply aren't equivalent forms of tensor-encoding that are as common/applicable as what you see in images. In this case, a strongly typed buffer of primitives dramatically simplifies questions of indexing and tensor-value access. This is the exact approach taken by the Arrow tensor spec (https://arrow.apache.org/docs/format/CanonicalExtensions.html#variable-shape-tensor). Again, most tensor libraries work under this assumption and so feeding a tensor library from a typed buffer + shape will be the most naturally way to work with this data.

### What about "RGB" Tensors?

All that said, it's still a very common pattern for a user to decode an image into an HxWxC (or CxHxW) tensor. And this is, in fact, what many users will expect to provide as an input. A numpy `ndarray` is a tensor -- not an image buffer.

Even for users working with images, whether the user expects to provide an Image (buffer + encoding) or a Tensor (ndarray) will heavily depend on where the user sits in the software stack of their organization.

Rather than fight against this, we may also want to support an "ImageTensor" archetype, which would be a Tensor datatype which we know stores the pixels of an image in one of the common tensor arrangements. This would not support any pixel-encoded images. Only those that had already been decoded into multi-channel tensors.

jleibs, 2024-05-27:
Most of the choices for working with tensors fall into one of 4 categories.

### Typed buffer, multiple data-types (the proposal)

Pros:
 - When processing a chunk the raw arrow data is much easier to work with
 - Opportunity to align with the official arrow spec for tensor representation
 - Aligns with our long-term direction of wanting to have multiple types and datatype conversions

Cons:
 - Multi-datatype representation means we must either proliferate typed components or introduce datatype conversions.

The current hypothesis is that proliferating types is a known challenge and can be mostly automated with a mixture of code-gen and some helper code, whereas datatype conversions is an unknown challenge.

Still this puts us on a pathway where once we support multi-typed components, we mostly delete a bunch of code and everything gets simpler. Any type conversions move from visualizer-space to data-query-space, but the types and arrow representations we work with don't actually need to change.

### Untyped buffer with type-id

Pros
- Avoids arrow unions while maintaining a single datatype.

Cons
- Forces arrow users to do annoying user-space datatype casting.
- Doesn't align with our long-term goals

### Typed buffer with union

Pros
- Status quo. Already works.

Cons
- Forces arrow users to do annoying poorly supported union operations when loading or reading tensors.

Wumpf, 2024-07-09:
How to handle `OutOfTreeTransform`?

Introduce a new boolean component that coerces a transform batch to be out of tree. `OutOfTreeTransform { enabled: bool }`
if one encounters several several _parent_ transforms and if they have the `OutOfTreeTransform == false` the viewer issues a warning and resolves in a best effort manner

To consider: The fallback provider for `OutOfTreeTransform ` can be clever and look at transform counts, automatically assuming `true` if there's several transforms.

Drawback: Can no longer have out of tree transform side by side with in-tree transforms