rerun-io/rerun #6403: Blueprint file from 0.15.0 isn't compatible with new 0.16.0 version

rgolovanov, 2024-05-22:
**Describe the bug**
Apparently the forward compatibility is broken between 0.15.0 -> 0.16.0 for blueprint files.

**To Reproduce**
Steps to reproduce the behavior:
1. Save blueprint in 0.15.0
2. Run viewer v0.16.0
3. Try to open saved blueprint
4. See error

**Expected behavior**
No errors. Loading older blueprint files should work.

**Screenshots**
[image]

**Desktop (please complete the following information):**
windows

**Rerun version**
0.16.0

Wumpf, 2024-05-22:
Unfortunately, rrd (recording) and rbl (blueprint) files are generally not expected to be backwards compatible at this point, just as the warning notes.
For that we'll need to establish a conversion process from version to version and enforce this etc.. At this point we're lacking this mechanism and are still breaking the formats too often at a more fundamental level. Reaching backwards compatibility for our formats is naturally a thing we want to reach at some point but we haven't determined yet when we want to start this.

Going to close this ticket since we'll not fix the particular 0.15 -> 0.16 issue.

closed 2024-05-22 by Wumpf

rgolovanov, 2024-05-22:
@Wumpf , thanks for clarification. yes, sure but the problem is general and probably, this issue can be converted to more generic feature request?

Wumpf, 2024-05-22:
👍  Done here now https://github.com/rerun-io/rerun/issues/6410