rerun-io/rerun #6410: Backwards compatibility of rrd & rbl (and mechanisms to ensure it)

Wumpf, 2024-05-22:
We'll need to establish a conversion process from version to version and enforce this. At this point we're lacking this mechanism and are still breaking the formats too often at a more fundamental level which is why we report incompatibility _even_ if they're compatible.

As of writing we haven't determined yet when we want to start this as it may incur significant costs to ongoing development while we're still trying to figure out what these formats are going to look like more long term.

emilk, 2024-10-31:
Rough roadmap:
* Replace our use of serde/msg-pack with protobuf (perhaps based on our gRPC protocol?)
* Add some sort of version-field to our component meta-data (e.g. as part of tagged components)
* Version our components (and archetypes?) in different modules (v0, v1, …) and keep the deserializers for the old ones, and `impl From<v1::Foo> for v2::Foo`
* Commit to having our enums and structs be additive (only add new fields and deprecate old ones, never change existing ones)

TKaluza, 2024-12-02:
Alternatives to consider:
https://capnproto.org/news/2014-06-17-capnproto-flatbuffers-sbe.html
https://dwrensha.github.io/capnproto-rust/2024/09/17/0.20-release.html
https://flatbuffers.dev/flatbuffers_guide_use_rust.html

closed 2025-04-24 by emilk