rerun-io/rerun #6411: In-viewer test runner

abey79, 2024-05-22:
It would be nice to have the ability to run automated tests with the full context of a fully running viewer (e.g. with `ViewerContext`, `ViewportBlueprint`, etc.) and known data. Given the amount of context needed, the viewer must basically be running and test executed from there.

The `antoine/in-viewer-test-runner` branch has a prototype of that, including:
- A registry of tests, to be triggered when a specific entity (`/test_trigger`) contains a specific component (`TestTriggerIndicator`), which contains the given test name as payload.
- A python helper along with a collection of python "test data" script, which logs some data along with the `TestTriggerIndicator` component.
Running the latter triggers the tests registered in the former.

Issues/notes:
- At the very least, `ViewportBlueprint` is needed for the tests. So #5421 is a prerequisite.
- Even then, that means that anything "below" `re_viewport_blueprint` would not have access to this test harness.
- Alternatively, one could possibly devise some kind of distributed tests runners, each local to the subsystems to be tested and with all the "local" context available. A shared "trigger orchestrator" would send messages to the runners to signal which test must be run when.

emilk, 2024-05-23:
For headless, we could simply make `ViewerContext::render_ctx` and `Option`. Then we could write unit-tests and benchmarks for almost all of our crates.

abey79, 2024-05-23:
Agreed. This would probably make the effort in this issue's description redundant.

IIRC, one soft-blocking thing is the ability to crate a store using nice SDK APIs, such that these unit test are not too horrible to write.

abey79, 2024-06-11:
Agreed.

For posterity, here is the rough prototype of this, should we want to revive it at some point: https://github.com/rerun-io/rerun/tree/antoine/in-viewer-test-runner

closed 2024-06-11 by abey79