rerun-io/rerun #6418: Collapsing ListItem button feels dead

emilk, 2024-05-23:
Two problems
A) only the arrow is clickable
B) the arrow does not respond to hover

[image]

Maybe A) is fine (it's consistent when clicking the title does something else), but B) is not.
Everything that is clickable should light up when hovered.

emilk, 2024-08-05:
B) is fixed, but A) is still an issue

abey79, 2024-08-05:
When the item is selectable (like in the blueprint tree), I think you dont want the "select" action (aka click) to trigger an unwanted change of collapsedness. This is the reason why clicking on the triangle specifically is required. We _could_ make the entire item clickable to un/collapse when `interactive(false)` though, but it would be somewhat inconsistant.

emilk, 2024-10-21:
@gavrelina what do you think about this?

gavrelina, 2024-10-21:
TLTR: Yes! 

I think it's a very good practice to make the whole label actionable to whatever action it belongs to, as it improves the affordance (in the case of the issue: open/close through the label, other example: check box is ticked even if user clicks on the label). Good comment from Antoine that we have sometimes combinations of those. In that case we should respect the action that is "the most contextual to the label". 

And the comment that it's not consistent —> I think that is not a problem, as it overall gives a better experience, and one may even say consistent in the context (even if not in implementation). Also! The section selections do close/open when one clicks on the label, so we will actually be more consistent! ;)  

I could not quickly find an example of combination that Antoine is mentioning, let me know a concrete example if someone sees one!