rerun-io/rerun #6487: Intermittent SIGBUS when spawning viewer from python

jleibs, 2024-06-03:
A user reported an intermittent crash when spawning the viewer from python on 0.16:
```
Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 
Report bugs: https://github.com/rerun-io/rerun/issues 

[2024-06-02T22:05:44Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-06-02T22:05:44Z INFO  re_sdk_comms::server] New SDK client connected: 127.0.0.1:58043
[2024-06-02T22:05:44Z INFO  re_sdk_comms::server] New SDK client connected: 127.0.0.1:58044
   3: re_crash_handler::install_signal_handler::signal_handler
   4: __simple_esappend
   5: std::sys::common::thread_local::fast_local::Key<T>::try_initialize
   6: crossbeam_deque::deque::Stealer<T>::steal
   7: rayon_core::registry::WorkerThread::wait_until_cold
   8: rayon_core::registry::ThreadBuilder::run

Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 
Report bugs: https://github.com/rerun-io/rerun/issues 
```

weidenka, 2024-06-04:
Some here on a Silicon Mac with MacOS 14.5 and `rerun-sdk-0.16.1`.
 
```shell
Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   3: re_crash_handler::install_signal_handler::signal_handler
   4: __simple_esappend
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: eframe::native::app_icon::AppTitleIconSetter::update
  17: <eframe::native::wgpu_integration::WgpuWinitApp as eframe::native::winit_integration::WinitApp>::run_ui_and_paint
  18: eframe::native::run::run_and_return::{{closure}}

Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
```

SeaOtocinclus, 2024-06-04:
Got another one (rerun viewer was not open)

```
Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 
Report bugs: https://github.com/rerun-io/rerun/issues 

   3: re_crash_handler::install_signal_handler::signal_handler
   4: __simple_esappend
   5: crossbeam_epoch::sync::once_lock::OnceLock<T>::initialize
   6: crossbeam_epoch::sync::once_lock::OnceLock<T>::initialize

Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 
Report bugs: https://github.com/rerun-io/rerun/issues 
```

Wumpf, 2025-03-10:
Another report of this here
https://discord.com/channels/1062300748202921994/1347915133095252039/1347915133095252039

lucasmerlin, 2025-06-17:
> we should revisit how this is implemented over on egui repo. Did anyone there report this issue as well?

I think this could be the same problem: https://github.com/emilk/egui/issues/7155

Wumpf, 2025-06-17:
@lucasmerlin nice!! seems very very likely. that one has a way better stack trace -  `NSImage` initialization is wrong somehow in that one

k-ai0, 2025-06-17:
@Wumpf 

MacOS Version: `15.4.1 (24E263)`

Pre-brew uninstall libpng, ran in terminal:
  `rerun` -> Yields SIGBUS bug
  `sudo rerun` -> Works as intended, no SIGBUS bug; why I suspect this may be a permissions thing
Post-brew uninstall libpng:
  `rerun` -> Works as intended, no SIGBUS bug

Extra brew notes:
```zsh
brew uninstall --ignore-dependencies libpng                                                                                
Uninstalling /opt/homebrew/Cellar/libpng/1.6.48... (28 files, 1.3MB)
libpng 1.6.47 is still installed.
To remove all versions, run:
  brew uninstall --force libpng
```

@Wumpf how does one do this?

>  If you want you can use the Rerun web viewer as a workaround meanwhile.

AFAIK I run `rerun` via calling `rerun`; my knowledge about this tool is nascent and am exploring if it can help to do image/video annotation and labeling like drawing polygons/boxes/etc.

Wumpf, 2025-06-17:
> Post-brew uninstall libpng:
> rerun -> Works as intended, no SIGBUS bug

awesome. Thanks for confirming. Still working on a better fix, see previously linked egui issue, but I have reason to believe that a lot of applications on Mac can be affected by this. Curiously this doesn't reproduce on all machines the some, there's some OS versioning or setup details at play as well.

> how does one do this? [starting the web viewer]

You can do so by `rerun --serve-web` but typically you'd do this from your own code via the SDK. Now that you do have the native viewer running I'd recommend to stick with it whenever possible, it's a lot snappier.
For details check https://rerun.io/docs/reference/sdk/operating-modes

> AFAIK I run rerun via calling rerun; my knowledge about this tool is nascent and am exploring if it can help to do image/video annotation and labeling like drawing polygons/boxes/etc.

Then welcome! Have a browse at the [examples](https://rerun.io/examples) and [snippets](https://github.com/rerun-io/rerun/blob/latest/docs/snippets/INDEX.md), hope you'll find what you're looking for.
We also have a [Discord community](https://discord.gg/pe9gakddfR) for any questions you might have

k-ai0, 2025-06-17:
Indeed, I appreciate the response. Thank you @Wumpf!

Wumpf, 2025-06-24:
There were other crashes mentioned in this thread, but since the Mac SIGBUS on startup is the only one we got proper hold off I'm going to close this as part of
* https://github.com/rerun-io/rerun/pull/10334

New issues should be opened for the other ones.

@jleibs @SeaOtocinclus did you ever see the `crossbeam_deque` related crash again? (or any other you assumed to be captured by this issue)