rerun-io/rerun #6532: API / dataloader for logging a video file directly

jleibs, 2024-06-10:
It is common for users to have video files they want to insert into Rerun.

Eventually, we would like to support video compression directly in the viewer (https://github.com/rerun-io/rerun/issues/5815), but doing so in a cross-platform way (including the browser) presents several challenges.

However, at present, we require users to decode video themselves and then log the individual image frames to Rerun. This is challenging to do permanently, generally requiring dependencies on external libraries, and often times multi-threaded/multiprocessing approaches to help with throughput.

It would be nice to offer a user-space API that allowed you to directly submit a file-path or video-encoded bytestream and let Rerun take care of converting to a stream of images. 

Although our data-loader APIs make it possible to do this as a plugin, it might be worth offering a dedicated API for this. In particular, a dedicated API would allow us to be explicit about how frame-ids map to time-points, which is not something we currently have the ability to do through data-loaders.

This is a situation where it would likely make sense to do the implementation in Rust once and then offer bindings through to the other languages.

emilk, 2024-08-27:
* Ongoing work in https://github.com/rerun-io/rerun/pull/7261

jleibs, 2024-09-10:
Needs to produce video frame references

closed 2024-09-17 by Wumpf