rerun-io/rerun #6552: Simplify blueprint GC for query-time clears

jleibs, 2024-06-12:
The blueprint currently depends on cleared data getting fully removed from the entity tree.

This is because we do certain entity-tree-walk operations that we don't want to grow in an unbounded fashion  as we add and remove space-views in a long-lived blueprint.

Making this work with query-time GC adds a lot of complexity for something we only depend on in the blueprint.

The proposal is to introduce an explicit store API for removing an entity_path in it's entirety. When you delete a space-view, we'll use this new API to fully remove it from the blueprint and bypass the need to GC to do this magically.

If we want to preserve rewind/playback functionality in the blueprint, we'll need the blueprint to use a historical latest-at-query to decide which things it purges from the store.

Wumpf, 2024-06-13:
I think this proposal would solve this issue, right?
* #1329

teh-cmc, 2024-06-13:
Yep

closed 2024-08-09 by jleibs