rerun-io/rerun #6557: Find a way to make common defaults more visible

Wumpf, 2024-06-13:
We should have a way to suggest changing certain component defaults per View.

Certain components are much more common candidates for overrides:
[image]

The tricky part is that we can't show a "current value" in all cases since generally, the current value is dependent on the exact dataresult (e.g. color fallback is implemented per visualizer and typically entity path dependent).

-> Either we limit ourself to components where we rely strictly on the placeholder (how to enforce?) or we somehow indicate that the default was not yet set.
The later is generally preferable since we want to be able to express what data is actually in the blueprint store, i.e. what blueprint has been sent from the code. If we were to show properties as-if a default was set, this makes the blueprint-code to viewer relationship quickly confusing!

jleibs, 2024-06-24:
This is addressed by the new override UI: #6605

Wumpf, 2024-06-28:
closing until we think it's not visible enough

closed 2024-06-28 by Wumpf