rerun-io/rerun #6572: Chunk metadata should be passed as an outer container

teh-cmc, 2024-06-14:
_Opening this right now since I need some place to link to for my TODOs in #6570_

We're currently using the native Arrow schema metadata feature to carry chunk metadata around.

This is way too limited, slow and cumbersome of a system for our needs (basically just untyped string tuples), and we're already hitting its limits.
We need a proper encoding for these things -- the usual solution is to use a row-based outer container such as a protobuf.

emilk, 2025-02-25:
Are we sure about this?

---

> too limited

We can use JSON to encode arbitrary data in the value fields. This is the recommended way for arrow extension types: https://arrow.apache.org/docs/format/CanonicalExtensions.html

---

> slow

Isn't that mostly mitigated by having large record batches?

---

> we're already hitting its limits

Where?

teh-cmc, 2025-02-25:
Discussed IRL.

TL;DR: Yes, I still think we need this as we start building internal data pipelines that ship a lot of chunks and their statistics around. In the short to mid term though, we have much more important stuff to think about.

teh-cmc, 2025-07-11:
* See also https://github.com/rerun-io/dataplatform/issues/950