rerun-io/rerun #6624: Changing color on a mesh doesn't have any effect

Wumpf, 2024-06-24:
This happens likely because we don't update the mesh in the mesh cache, so this likely applies to all properties of the mesh, but color is the only one where editing via ui makes sense.
(so if you were to override positions and send that to the viewer from a python repl, it would likely suffer the same issue)

Wumpf, 2024-06-25:
(as expected:) now that I made material editable, we have the same problem there as well (on a branch as of writing)

abey79, 2024-06-26:
Editing most components of mesh seems to have no effect. Should we have some per-archetype component exclusion list to just hide stuff from the UI?

Wumpf, 2024-06-26:
better, we fix the issue. What needs to be done is that we use all rowids of whatever source the data is from is used as the cache key for the mesh

commit 65e64744 (2024-06-27): Introduce new  query result hash and use it to fix mesh cache keying (#6671)

### What

* Fixes #6624

<img width="2406" alt="image"
src="https://github.com/rerun-io/rerun/assets/1220815/60a5b8ec-9ffb-4115-9b5a-7fe6aa967c3b">

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [ ] I have tested the web demo (if applicable):
* Using examples from latest `main` build:
[rerun.i [... cut]
closed 2024-06-27 by Wumpf