rerun-io/rerun #6642: Viewer instances don't release memory when cell output is cleared

jleibs, 2024-06-25:
A given notebook session can eventually run out of memory at which point the only way to fix it appears to be reloading the page.

My expectation is that when the output of a cell is cleared, the memory should be freed.

To repro, repeatedly execute a cell that outputs a viewer widget until it stops outputting.

At this point the console output for the page will show:
```
WebAssembly module validated with warning: failed to allocate executable memory for module
Uncaught (in promise) InternalError: out of memory 
```
[image]

It would be nice if we could also catch this and show the error in the notebook. Instead all we get is an empty white canvas and the newly introduced timeout warning.
[image]

jleibs, 2024-06-25:
I noticed that every time I re-execute a cell, I another instance of WebAssembly.instantiate being tracked:
[image]

emilk, 2024-06-26:
Is this a new problem, or a pre-existing one? Not that it matters a lot, seems very high-priority to fix it

jleibs, 2024-06-26:
It's new. The old notebook approach added an iframe to the page. When the previous iframe is removed from the DOM, it seems like the right cleanup happens.

jleibs, 2024-06-26:
Digging into the memory inspector and taking snapshots after each cell execution, you can see we leak approximately 100MB per execution. Something is holding a reference that keeps it from being GC'd.

[image]

jleibs, 2024-06-26:
I believe this is actually an upstream anywidget issue: https://github.com/manzt/anywidget/issues/613

closed 2024-07-01 by emilk