rerun-io/rerun #6675: C++ Transform3D() archetype constructor should match rust Transform3D::IDENTITY

jleibs, 2024-06-27:
In <TODO>, the roundtrip test between:

```cpp
 auto base_axes = rerun::Transform3D(rerun::datatypes::TranslationRotationScale3D::IDENTITY)
                         .with_axis_length(1.0);
```
and
```rs
let base_axes = rerun::Transform3D::IDENTITY.with_axis_length(1.0);
```

Fails with:
```
Error: Found discrepancy in row #0: union arrays' type indices don't match!
Differences (-row1|+row2):
-Row #17DD008D0CBAABAD7DE6842282C89594 @ '/base'
+Row #17DD008D983257ED5D001D2DEAD3CBDC @ '/base'
 - log_tick: #0
-┌───────────────────────────────────────────────────────────┐
-│ CHUNK METADATA:                                           │
-│                                                           │
-├╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┤
-│ ┌─────────────┬──────────────────┬──────────────────────┐ │
-│ │ AxisLength  ┆ Transform3D      ┆ Transform3DIndicator │ │
-│ │ ---         ┆ ---              ┆ ---                  │ │
-│ │ type: "f32" ┆ type: "union[3]" ┆ type: "null"         │ │
-│ ╞═════════════╪══════════════════╪══════════════════════╡ │
-│ │ [1]         ┆ [-]              ┆ [-]                  │ │
-│ └─────────────┴──────────────────┴──────────────────────┘ │
-└───────────────────────────────────────────────────────────┘
+┌────────────────────────────────────────────────────────────────────────────────────────────────────────┐
+│ CHUNK METADATA:                                                                                        │
+│                                                                                                        │
+├╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┤
+│ ┌─────────────┬───────────────────────────────────────────────────────────────┬──────────────────────┐ │
+│ │ AxisLength  ┆ Transform3D                                                   ┆ Transform3DIndicator │ │
+│ │ ---         ┆ ---                                                           ┆ ---                  │ │
+│ │ type: "f32" ┆ type: "union[3]"                                              ┆ type: "null"         │ │
+│ ╞═════════════╪═══════════════════════════════════════════════════════════════╪══════════════════════╡ │
+│ │ [1]         ┆ [{translation: -, rotation: -, scale: -, from_parent: false}] ┆ [-]                  │ │
+│ └─────────────┴───────────────────────────────────────────────────────────────┴──────────────────────┘ │
+└────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

We can work around this by instead using:
```
 auto base_axes = rerun::Transform3D(rerun::datatypes::TranslationRotationScale3D::IDENTITY)
                         .with_axis_length(1.0);
```
which is quite annoying.

jleibs, 2024-06-27:
I tried to add a new `Transform3D()` constructor via codegen, but the generated `= default` constructor gets in the way.

Wumpf, 2024-06-28:
why not add `rerun::Transform3D::IDENTITY` to C++ as well 🤷

Wumpf, 2024-07-18:
(more shortterm the default ctor of Transform3D is now already giving us an empty Transform)

closed 2024-07-18 by Wumpf