rerun-io/rerun #6681: Sending a blueprint withan empty barchart crashes the viewer

jleibs, 2024-06-29:
Repro:
```
import rerun.blueprint as rrb
import rerun as rr

rr.init("rerun_example_crash", spawn=True)

uhoh = rrb.Blueprint(
    rrb.BarChartView(contents=["fft"]),
)

rr.send_blueprint(uhoh)
```

jleibs, 2024-06-29:
Things to investigate:
 - egui_plots has been updated
 - aspects of fallback values have been changed that could cause us to be sending different data to egui_plots in this case.

Wumpf, 2024-06-29:
that's a debug assertion only, fixed it for the time series chart already but didn't realize there it was new

emilk, 2024-06-30:
We also hit a `debug_assertion` just loading the plots example

emilk, 2024-06-30:
>  fixed it for the time series chart already but didn't realize there it was new

@Wumpf is referring to https://github.com/rerun-io/rerun/pull/6639/files

I don't really understand why `scalar_axis.component_or_fallback` returns an inverted range though

emilk, 2024-06-30:
I believe this has something to do with empty plots, and how to pick bounds for such plots

Wumpf, 2024-06-30:
Doesn't have to be inverted, a zero size range is enough to trigger the assertion. (A user may though provide an inverted range, we have to guard for that as well)

emilk, 2024-06-30:
When I run `check_all_components_ui.py`, `scalar_axis.component_or_fallback::<Range1D>` return the range `(3.0000000000000004, 3.0)` - I wan't to understand why.

emilk, 2024-06-30:
I suspect this is to blame: https://github.com/rerun-io/rerun/blob/9c4df0b8abd62929915ded783e377285a716ac0c/crates/re_space_view_time_series/src/space_view_class.rs#L665

closed 2024-06-30 by emilk