rerun-io/rerun #6685: Periodic calls to rr.save() is not saving anything

vmayoral, 2024-07-01:
Since my test setup logs quite a bit of information, i started testing out periodic saves implemented as:

```bash

rr.init()
rr.serve()
while True:
  # do some stuff
  rr.save()
  rr.serve()  
```

The resulting saved files contain nothing useful:
```bash

-rw-rw-r-- 1 victor victor 279 Jul  1 07:28 _20240701_072815.rrd
-rw-rw-r-- 1 victor victor 279 Jul  1 07:28 _20240701_072825.rrd
-rw-rw-r-- 1 victor victor 279 Jul  1 07:28 _20240701_072835.rrd
-rw-rw-r-- 1 victor victor 279 Jul  1 07:28 _20240701_072845.rrd

```

teh-cmc, 2024-07-01:
You cannot `serve` and `save` at the same time since these are both stateful operations that swap the underlying sink.

I have never tried it, but doing periodic calls to `save` with different filepaths without `serve` calls intersperse inbetween should actually work, I think? Curious to hear back from you on that one.

The ability to log data to multiple sinks at once is on our roadmap but I don't have an ETA quite yet:
- https://github.com/rerun-io/rerun/issues/1998

vmayoral, 2024-07-02:
> I have never tried it, but doing periodic calls to save with different filepaths without serve calls intersperse inbetween should actually work, I think? Curious to hear back from you on that one.

Thanks for the answer. I can confirm that this works out. Commenting out the serve() and keeping the periodic `save` calls works fine. Then, all files can be loaded with `rerun /path/to/files/*.rrd`. Will continue keeping an eye on https://github.com/rerun-io/rerun/issues/1790 to do things programatically then.

Will also be keeping an eye on https://github.com/rerun-io/rerun/issues/1998.

closed 2024-07-02 by vmayoral